huizebruin/s0tool #19: When using s0tool/esphome/watermeter.yaml it reports every pulse as 3.6L instead of 1.0L

compacthack, 2022-04-16:
Hello, I'm using your code for measuring water usage and it works great, and it works well with the update interval as 1s instead of default 60s, but it reports every pulse from sensor as 3.6L instead of 1.0L
I'm sure its a function of reducing the update interval to 1s, I played with the code and when it is set to count pulses over 60s & removed the filter it reports correctly, but you only get updates every minute and the update lag is annoying. 
Maybe if the reported value is just divided by 3.6 then it should report the pulse as 1L. 
I was also thinking of trying the esphome pulse count - rising_edge option, and triggering as the water meters spinning flag 1st triggers the sensor, as the flag may come around and trigger the sensor but stops there and won't record 1L usage until the flag as passed and the pulse finishes ?
Thanks for all your work.

huizebruin, 2022-04-16:
Would you like to try this one out for me?

https://github.com/huizebruin/s0tool/blob/dev/esphome/pulscounter.yaml

I made some changes.

Wobbe

compacthack, 2022-04-16:
Hi Wobbe, nice work. That is working great now. Will test over a couple of Days. Thanks for your hard work

compacthack, 2022-04-17:
I may have spoken to soon, the id: "sensor_pulse_meter_total" is showing 1L correctly every pulse, but "id: watermeter_total" goes up by 6L every pulse. Another question and this might be more of a HA question but is there anyway to manually change the "id: watermeter_total" reading so it starts counting from a particular number, then it would be calibrated with the real reading from the physical water meter? thanks again. 

[image]
 
[image]

huizebruin, 2022-04-19:
How is it now ?

closed 2022-05-05 by huizebruin