huizebruin/s0tool #35: Sensor watermeter stand

ydeschoe, 2022-09-06:
After each restart of the s0tool my watermeter stand goes always back to the following value :201.000m3

how will it keep the last value on a reboot of the sensor. 

regards

yves

huizebruin, 2022-09-07:
I'm working on it, 
That's 1 of the first actions that I'm working on.

If someone have a direction how to do it please let me know.

huizebruin, 2022-09-10:
Hello yves,

I made a couple of changes would you try it out? 
woud you try the stool complete (standard) to program on your device
[https://huizebruin.github.io/s0tool/](https://huizebruin.github.io/s0tool/)

thomasvt1, 2022-10-29:
I think it's still an issue @huizebruin, the global variable is only saved in esphome when `globals.set` is called (which isn't done).

A block like this should work
```
      on_value:
        - globals.set:
            id: totalWaterUsage
            value: id(sensor_pulse_meter_total).state + id(totalWaterUsage); 
```

Altough it might be an idea to add another global float with initialWaterUsage, which you set once using the service and then combine it with the totalWaterUsage in a template. `initialWaterUsage + totalWaterUsage = currentMeterValue`.

I might be making it a bit too complex, so please simplify if needed!

huizebruin, 2022-10-29:
Hello Thomas,

I will test it out on my own setup.
The first one.

I'm just a beginner at esphome and programming. I'm working on this now from the beginning of this year 2022.

And have no experience with programming languages.

But the s0tool code is a beginning of a nice thing to learn.

thomasvt1, 2022-10-29:
As a non-programmer the code looks very clean, so well done!

I did a bit of testing and I think the solution would look something like this;
https://github.com/thomasvt1/s0tool/blob/main/esphome/components/watermeter.yaml - I haven't done any validation at all but theoretically it should work

huizebruin, 2022-10-30:
> After each restart of the s0tool my watermeter stand goes always back to the following value :201.000m3
> 
> how will it keep the last value on a reboot of the sensor.
> 
> regards
> 
> yves

Thanks to @thomasvt1 i made a new version.
i will test it tomorrow if its good i wil post it.
Its holding the meter nummers after update and power loss. 

wobbe

huizebruin, 2022-11-01:
> After each restart of the s0tool my watermeter stand goes always back to the following value :201.000m3
> 
> how will it keep the last value on a reboot of the sensor.
> 
> regards
> 
> yves

@ydeschoe can u test it, its now working fine here, thanks to @thomasvt1

closed 2022-11-12 by huizebruin

array81, 2023-07-07:
Who updates meter_value variable?
I can upgrade it by Home Assistant but EspHome didn't upgrade, right?