huizebruin/s0tool #70: Sensor watermer_totaal too high?

Nakebod, 2022-11-12:
Hi,

Recently I came across your script, awesome project!
So I ditched my old selfmade config, which didn't integrate with the Energy dashboard anyway. But it was functional.
As hardware I have an ESP8266 with NPN sensor

To track my daily water usage I have a custom utility meter in configuration.yaml:

`````yaml
utility_meter:
  waterverbruik_per_dag:
    #source: sensor.watermeter_totaal
    source: sensor.watermeter_pulses_totaal
    cycle: daily
`````
As source I tried using the watermer_totaal, but those values seems to be too high.
It also increases while there is no water usage, 100-150 liter last night is almost more than my real usage during the whole day.
Since I use sensor.watermeter_pulses_totaal as source it is accurate again.

So there is a big difference in values. 
My daily meter is currently: 36 liter vs 129 liter from the watermeter_totaal sensor. Since the HA Energy dashboard is also using the totaal_sensor as source the values there are also way too high.
Okay one sensor is in liters, the other in m³, but that shouldn't cause a 3.5 times bigger value. Unless 100 liter isn't 0.1 m³ anymore :)

The sensor.watermeter_stand looks very accurate.

huizebruin, 2022-11-12:
Hello @Nakebod .

thanks for the question.
i was asked the same from someone from my website.
and also by mail. I think i found it and also the other problems whit the Counter when u update it from Home Assistant. like this one. i am testing is right now. 

https://github.com/huizebruin/s0tool/issues/64 also its fixing this one
https://github.com/huizebruin/s0tool/issues/35 also its fixing this one

Nakebod, 2022-11-12:
Ah saw those issues earlier, but didn't think they where related.
Will wait for your fix. Thanks!

closed 2022-11-12 by huizebruin