huizebruin/s0tool #77: Help with Maddalena CD One TRP

wes1993, 2022-11-21:
Hello, 
i hope someone could help me and this is the correct way to post:

I have the Maddalena CD One TRP water meter and i have bought this sensor from Aliexpress:

https://it.aliexpress.com/item/32582384960.html?spm=a2g0o.new_account_index.0.0.7b6825b9A0W8v0&gatewayAdapt=glo2ita

I have installed and configured everything correctly infact the sensor works well when i try with a magnet but if i put on the water meter won't work... do you have some suggestion on how i can test if my water meter magnet work well?

I have seen form the Maddalena website that my water meter is compatible with this sensor:

https://www.maddalena.it/prodotto/reed-switch-trp-one/#form-product

I hope in you reply
Bye
Stefano

HobbitBluelabel, 2022-11-21:
Hi Stefano,
Are you sure that 5 mm detection  is enough? 
I need a 8mm sensor on my watermeter to get correct pulses...
Bye,
Jan-Willem

wes1993, 2022-11-21:
Thanks a lot @HobbitBluelabel for your fast reply :-D
I don't know that there is difference between sensors apart from the voltage...
I think i need a 8mm sensor at this point... there is a way to test the mm?

Bye
Stefano

wes1993, 2022-11-21:
@HobbitBluelabel 
P.S. can you send me the link of your sensor?

HobbitBluelabel, 2022-11-21:
Heschen inductieve nabijheidssensor schakelaar LJ18A3-8-Z/AX detector 8 mm
6-36 VDC 300mA NPN normaal gesloten(NC) 3 draad https://amzn.eu/d/bBJkKuf

huizebruin, 2022-11-22:
Npn Sensor [Aliexpres](https://s.click.aliexpress.com/e/_AVaoGr)

that one is 8mm 
M18 8mm sensing DC 5V inductieve proximity sensor switch NPN GEEN LJ18A3-8-Z/BX-5V

huizebruin, 2022-11-30:
> Hello, i hope someone could help me and this is the correct way to post:
>
> I have the Maddalena CD One TRP water meter and i have bought this sensor from Aliexpress: [...]

did u fixed it ?

wes1993, 2022-11-30:
I have ordered the new sensor from aliexpress, when arrive I will do the test and keep you and the post updated :)

thanks a lot

HobbitBluelabel, 2022-12-01:
Thanks!
Hartelijke groet,
Jan-Willem van den Hoek

wes1993, 2022-12-06:
Hello @HobbitBluelabel,
Today I have received the new sensor, 8mm distance, I have installed everything again but unfortunately this sensor, like the other, won’t work… if I try with piece of iron works well…. So at this point I don’t know what is the best sensor for my meter… probably I need another more performant sensor connect with an optocouper…

Someone have suggestion?

huizebruin, 2022-12-07:
[image]

I found this picture on Google.
Did you try it in that section?

wes1993, 2022-12-07:
Yes, I have tried but unfortunately won’t work… I don’t know why… I thin 8mm are not so much…

huizebruin, 2022-12-25:
> Yes, I have tried but unfortunately won’t work… I don’t know why… I thin 8mm are not so much…

Do u have a picture for me from your meter?
type and with the counters

wes1993, 2023-02-02:
@huizebruin,
sorry for my very late response.. :-( unfortunately i have read only tosay that you have replied to this...
Here is my meter (Maddalena CD One TRP):
[image]
[image]

Bye
Stefano

wes1993, 2023-02-02:
Unfortunately for now I can’t find a sensor that works.. I have bought 4mm and 8mm but seems that won’t work… :(

wes1993, 2023-02-04:
I have found a solution to this (It can be useful also for tracking single device, like washer machine etc.):
I have bought this sensor:
https://www.amazon.it/dp/B08KGG1H8Y?psc=1&ref=ppx_yo2ov_dt_b_product_details

Then i have connected the cable in this way (without Resistor):
Sensor --> ESP
BLK --> GND
RED --> +5V
Yellow --> D2

This configuration works well and i can read correclty the pulses but... the l/min value is not correct... this is what is written in the page:
[image]

Can we add a custom value instead of the select menu so we can use custom values to know the l/min?

I have seen in the code that s0tool do this operation for read the l/min:

return x * atof(id(Select_pulse_water).state.c_str()) * 1000

Probably. what i can insert the correct value i can calculate the correct value to insert.

Thanks again a lot for your wonderful work.
Stefano

wes1993, 2023-02-04:
This is the value that the program read, we know that this is not possible :-D

[image]

I have only opened a faucet not a hydrant :-D

huizebruin, 2023-02-05:
do you see the pulsrate water ?
there you can adjust the counting ist normaly 0.001 (1 puls a liter)
if needed i can make it more adjustable

wes1993, 2023-02-05:
Hello @huizebruin 
I can read the pulses because here it will change ON/OFF state:
[image]

But the problem is that from what i have seen in the specs of this sensor 5 pulses is 1 l/min

If you want i can test if you make a new release 

Do you have discord?

Bye 
Stefano

wes1993, 2023-02-05:
P.S. Hopw did you calculate the l/min ( you do a sum of old value or something else?

huizebruin, 2023-02-05:
yes i have discord, and have a Dutch server for my work, but have now also a English channel about the s0tool ( and his code) [https://discord.gg/k9cRBSNr](https://discord.gg/k9cRBSNr)

So871, 2023-02-16:
> @huizebruin,
> sorry for my very late response.. :-( unfortunately i have read only tosay that you have replied to this...
> Here is my meter (Maddalena CD One TRP):
> [image]
> [image]
> 
> Bye
> Stefano

It looks like you don't have a metal part on the meter that can be picked up by the NPN sensor.

The npn needs a metal part that triggers it, and judging from your picture it's not there.
So you won't be able to read it with a pulse sensor.

closed 2023-09-05 by wes1993