huizebruin/s0tool #78: Updated to v22.11.25: Watermeter stand has 4 digits after the comma

JGeeforce, 2022-11-24:
I just updated to v22.11.25. At first the watermeter stand went to the default value of 8000, which I didn't set. But after changing it via Services the meterstand has 4 digits after the comma, 1989,5350 for example. It should have 3 I guess. 

All values have 4 digits now......
[image]

huizebruin, 2022-11-24:
Why the Meter counter is set to 8000 I don't know at this point.

I have nothing set to that number.

And the 4 numbers after the , is because of the meter that's counts every 0.5l

But at this point I don't know how to make a easy switch between 3 and 4 after the comma.

We are working at this.

JGeeforce, 2022-11-24:
Second time the change went fine. Let's forget the 800 for now.

Thanks for the explanation. Hopefully it will be back to 3 (in my case) soon.

huizebruin, 2022-11-24:
i am going to push it now to the 3 numbers after the comma.

JGeeforce, 2022-11-24:
Thanks for the hard work.

closed 2022-11-24 by huizebruin