huizebruin/s0tool #85: yaml errors

gwittenaar, 2022-12-15:
First of all, very nice project.
But when i tried to adopt the device in esphome it came back with the message: failed config. See attached screen for details.
Probably i do something wrong but i don't what.

[image]

huizebruin, 2022-12-15:
You must update your esphome to minimal version 2022.11.x

And also have Home Assistant version v2022.11.x minimal.

Than it's must be going to work.

gwittenaar, 2022-12-15:
Thanks for the fast reaction.
You're right. I was using the version 2022.3.xx but i didn't noticed it was the community version, thats why i never got an update because you have to use the official version. So i installed the official version (2022.12.xx) and it works.
Thanks.

closed 2022-12-15 by gwittenaar