huizebruin/s0tool #95: Different pin configuration

laemen, 2023-01-03:
I use a D1 mini V4 esp8266 which has a different pin name. I used a manual config before, but switched to the automated version, now I am looking to where this config needs to go. Some is already there like id, but added just for reference. Looking for the pin definition and possibly the update interval.

esp8266:
  board: d1_mini
  
sensor:
  - platform: pulse_counter
    pin: GPIO2 # D2
    id: watermeter_pulse
    name: "watermeter pulse"
    state_class: measurement
    update_interval: 5s
    accuracy_decimals: 1
    unit_of_measurement: "l/min"
    icon: "mdi:water-pump"

huizebruin, 2023-01-03:
Hello @laemen 

esp8266 wemos d1 mini v4.
Does not have a 5v output at the same point as the v1 of the wemos d1 mini.
And u have 5v needed for the npn sensor.

laemen, 2023-01-03:
Yes I’m on discord, but I solved it by changing the hw to GPIO4 which is the D2 pin.
The GPIO2 I used (=D4) also has the led connected, so I could visually see if the esp had proper input, but according to [randomnerdtutorials](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/) the D4 pulled low could cause a failed boot.
For future reference having certain settings configurable through substitutions could help, but for now it’s solved.
S0tools is really easy, thanks for developing it!

closed 2023-01-03 by laemen

kanuva, 2023-03-29:
I was actually wondering if we can change the pinout in a more simple way?
The reason being is pin D2 (gpio4) is also used for part of the communication pins for I2c and I have a Wemos DHT12 shield laying around which I wanted to use. This shield uses I2c and I can't change the i2c pins because it's a shield made for the wemos d1 mini.

So what I've done right now is copied the git code and pasted into my own (public) github repo and altered the code a bit so now it uses d3 (gpio0). For now I'm only using my own repo to assigning the correct pin for the watermeter. But this feels a bit obsolete. Isn't there an easier way to do this?

huizebruin, 2023-04-03:
Hello kanuva,

Saidly i can not change it afterwards.
As i now of.

And i made a pcb design, with i send to people in Holland and Belgian. 
And that one is also using the d2 and d5 ports.

The only way is what u already did. To fork it and change the Pinout.