huizebruin/s0tool #96: sensor.totaal_opgebracht indicates higher then normal during higher pulsrates.

ThomasVerl, 2023-01-04:
Hi,
I have noticed that 'sensor.totaal_opgebracht' and related sensor readings per hour/day/week are not correct (too high) during higher power consumption (higher pulsrate). 
However, the readings for the 'sensor.actuele_meterstand' are accurate and correct.
Do you have any idea why or solution for this? 

Thanks

[image]

ThomasVerl, 2023-01-05:
> Hello @ThomasVerl<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FThomasVerl&data=05%7C01%7C%7Cbfa538ab6b48402a6e9008daee6d4166%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638084450265397910%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=3F9hWaIJHvKKNCmeztaxhbWQlSCAV5Z0nfu8JPdZfic%3D&reserved=0> .
>
> My question is what do u use as the S0tool pcb or did u build it yourself ?
> And can u give me the counter that u use and witch firmware u are using for the s0tool

Hi,
I built it myself and used the latest firmware v22.12.27 . The meter type is a DTS353<https://www.ytl-e.com/product/three-phase-smart-meter/YTLDTS3530.25-5(80)ADINrail3Phase7WEnergyMonitorMeterMIDCertified.html>.

huizebruin, 2023-01-05:
I found this, and knows that esphom puls is between 50 and 100, but i can make a version for that one, so u can test it out. if u like.

Impulse connection:
Voltage 12~27V, current ≤27mA, max length of wire:20m
Impulse width=90ms, connect to terminal 20&21
Limits of values: max 60VDC, max 50mA

https://www.kwhmeter.nl//Files/2/26000/26485/Attachments/Product/rY728An37477m00Q966983Q80ULo51VV.pdf

ThomasVerl, 2023-01-09:
Yes, that sounds great! As a temporary solution, I changed the sensor in the utility.yaml file by replacing it with sensor.actual_meter.

warmtepomp_test_nieuw:
    source: sensor.actuele_meterstand
    cycle: daily

huizebruin, 2023-01-19:
> Yes, that sounds great! As a temporary solution, I changed the sensor in the utility.yaml file by replacing it with sensor.actual_meter.
> 
> warmtepomp_test_nieuw: source: sensor.actuele_meterstand cycle: daily

I made some changes to the code, i will test is tomorrow.
when it's looks good i will upload it to github.
Also made a lite change so u can add the kWH meter stand to the energy dashboard.

huizebruin, 2023-01-21:
i made a update to V23.01.28 can u update to that one, i made some fixes

closed 2023-10-09 by huizebruin