huizebruin/s0tool #100: sensor sometimes registers usage while there is no water consumption

joost01, 2023-01-12:
I'm using your s0tool implementation for measuring my water usage in HA. This is an own build setup, using an ESP32 (demos) and a standard sensor. The setup was really a breeze, the project is well documented and the integration in HA is perfect.

However, I have problems with the measurement. I've searched on several forums but didn't find the answer. Sometimes, the meter registers high water usage while there is no consumption. I think it is related with the position where the measured metal plate stops. If this is at the sensor, it keeps counting. So I might have a water usage (according the sensor) of 12000 ltr a night. 

I'm not sure how the measurement is done. Does it count on "rise-up", "fall-down" or a combined action. The last option (combined) should be the most robust in my opinion.

Am I doing something wrong in this case?

joost01, 2023-01-12:
Hi Jan-Willem,

in my case it is the sensor / ESP. I have set the initial meter value. I just compared the actual meter value with the current total value in the webinterface of the ESP32, and in a few days it is over 3000 ltrs off. It counted 3000 ltrs more than I've actually consumed.

huizebruin, 2023-01-17:
The code is build for a esp8266, not for a esp32.
there are some changes need for to work on a esp32.

closed 2023-02-24 by github-actions[bot]