huizebruin/s0tool #101: Wemos D1 mini pro CP2104 not able to flash s0tool

MPJK77, 2023-01-19:
Not able to flash s0tool to my wemos D1 mini pro CP2104.
I can flash everything else on the wemos (e.g. WLED, Tasmota etc.)
I am not able to flash s0tool via the browser or via Home Assistant.
Is there any other way to flash s0tool?

huizebruin, 2023-01-21:
Fixed it with the new release.

huizebruin, 2023-01-22:
i made one special for the wemos D1 mini pro.
check the flash page under specials.
[https://huizebruin.github.io/s0tool/](https://huizebruin.github.io/s0tool/)

MPJK77, 2023-01-22:
Dag Wobbe, ik heb de special net geprobeert echter deze werkt helaas ook
niet. Said heeft hem ook geprobeert en ook hij krijgt hem niet geflashed.

Groeten Maarten (MPJK77)

vincenttor, 2023-02-21:
ik denk dat ik hier ook tegen aan loop, wat ik ook doe hij pakt hem niet.
Zal is met een ouder model chip proberen of deze wel gaat, wilde al een bericht sturen met een scherm opname dat er weinig tot niks gebeurd.

prvakt, 2023-03-14:
Having same issue with D1 mini pro CP2104 - flashing is stucked on "preparing installation" screen. Any chance this can be fixed?

huizebruin, 2023-03-16:
At this moment I can't find wy it is not working. When I look at the code I see the right device but it won't install for no reason.

huizebruin, 2023-06-10:
it wil only work on a wemos d1 mini v1 /V3 and v4.
the others will not work.

closed 2023-06-10 by huizebruin