huizebruin/s0tool #108: It looks like there is an issue with the calculation of the total water amount

MPJK77, 2023-02-01:
Hi,

It looks like the calculation of the total amount is wrong. (After updating ESPhome in HA a few days ago)
My usage would be extreme today wich is not the case.
Maybe there is an issue with dots and commas? (look at the different notation between pulserate and usage)
[image]
[image]
[image]

Also if you compare the increasing of "Stand" and "Totaal" it does not equally add up.

Regards, Maarten

HobbitBluelabel, 2023-02-01:
Did you add the watermeter to the energy dashboard?
Then your usage is totally wrong. I have the same problem. If i remove it
from the energy dashboard, everything is fine .

MPJK77, 2023-02-02:
I did remove the watermeter from the energy dashboard yesterday. And reset and restarted the device with no luck.
There is realy something wrong with the conversions I think.
[image]

This whas all set to zero yesterday at 22.10 p.m.
[image]

MPJK77, 2023-02-02:
I did update the divice and reset all sensors.
How do u explain a usage of 136990 m3 of water in 1 day?
And look at the peak of 5000 m3 this morning at 06.00 am how is that possible?

huizebruin, 2023-02-02:
i think that the npn sensor stands on the metal for a couple of time , and than counting on the numbers.
that's the only thin i can tink of.

So871, 2023-02-02:
MK , stel eens niet zo'n moeilijke vragen.....

But , There seemed to be a similar issue in esphome version 2022.10.0 .

@huizebruin you could try to test this with the fix issued in issue 3730.
 https://github.com/esphome/issues/issues/3730

closed 2023-06-17 by github-actions[bot]