huizebruin/s0tool #109: diy watermeter install from https://huizebruin.github.io/s0tool/

DoerakkerHA, 2023-02-09:
S0tool Watermeter | INSTALL | only watermeter
Install S0tool watermeter

na install verschijnt het device niet in ESPHome en visit device:
10:11:15 | [W] | [api.connection:085] | 192.168.178.53: Reading failed: BAD_INDICATOR errno=11

ik vermoed dat het een probleem is met een api key maar geen idee hoe te fixen
het device s0tool-s0-watermeter-d3234e is aanwezig en breikbaar o het netwerk  192.168.178.6

please advise

huizebruin, 2023-02-09:
Word hij wel geïnstalleerd op een WeMos d1 mini v1 ?

DoerakkerHA, 2023-02-10:
Goedemorgen Wobbe. Tot gisteren toen ik zelf de installatie in elkaar knutselde was er geen probleem. Op dit moment loopt HA vol met foutmeldingen; te veel om hier op de smartphone op te noemenIk zal straks even met een schone lei proberen te beginnen; misschien heb ik daar zelfs je hulp bij nodig (iets over platform esp home en unique IDs niet aanmaken sensoren)Kind RegardsROnald GroeneveldSent from my iPhone +31 650 989 689

huizebruin, 2023-02-15:
Wordt gesloten, probleem is opgelost.
hulp geboden op discord groep over de s0tool

closed 2023-02-15 by huizebruin

pietervanharen, 2023-03-06:
I have the same problem, but couldn't find the Discord group you mentioned. Do you mind posting the solution here or linking to the post in that Discord group?

huizebruin, 2023-03-06:
https://discord.gg/drnZRK2pq4