huizebruin/s0tool #119: how to assign static IP address to s0tool

AppleFlapp, 2023-04-05:
Hi, a noob here just starting with esp home.

I want to give the s0tool a static ip address but where can i do that?
I followed the installation guide but I was not able to change the IP address anywhere.

Thanks for your help.

PS. It can be done in Dutch but in English might reach more people.

huizebruin, 2023-04-06:
Did you put it under the code after you flashed it by the GitHub page?

AppleFlapp, 2023-04-07:
Yes I did. I just edited the YAML file via ESP home by adding a few lines.  Last five lines in the code below.

`wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
    # Optional manual IP
  manual_ip:
    static_ip: !secret static_ip_s0tool
    gateway: !secret gateway
    subnet: !secret subnet`

I did also add the line to set power save mode to none because I had connection losses every couple of minutes.
Still not convinced this is working properly now.

And we are getting at another issue now, I would like to get rid off the integrated webserver too
because it is not very friendly for the ESP8266 board according the ESP home website.
But it looks like I can't use the download option via the Github page then because another yaml file needs to be modified for that.

> Please note that enabling this component will take up a lot of memory and can lead to problems, especially on the ESP8266.

huizebruin, 2023-04-07:
> And we are getting at another issue now, I would like to get rid off the integrated webserver too
> because it is not very friendly for the ESP8266 board according the ESP home website.
> But it looks [...]

I'm using it for over a year and never had any problems with it.

HobbitBluelabel, 2023-04-07:
Same here....
Noticed no problems with the webinterface. Running stable for months.

Hartelijke groet,
Jan-Willem van den Hoek

closed 2023-05-16 by github-actions[bot]