huizebruin/s0tool #135: How to update

cyberfilou, 2023-10-16:
hello 

I have versie v23.09.32. and a want to update to V23.04.30
how should I do this without losing the existing data
THANKS
I use Home assistant last Version, and only S0Tools for water ( D2 )

Cyberfilou

huizebruin, 2023-10-23:
I'm working on it how te do the update.
on paper and when its done i will put it online

huizebruin, 2023-11-25:
[image]

Go to your ESPHome dashboard .
1. click on the 3 dots by the S0tool.
2. click on validate , so its getting the new code from GitHub.
3. click on install and wait until its finish.

Now you should have the latest version on it.

closed 2023-12-10 by huizebruin