huizebruin/s0tool #138: downside of ESP32?

bkbartk, 2023-11-25:
Hello 
I'm planning to change the wemos d1 to a wemos d1 ESP32.
The main reason is the wifi performance.
The pins are exactly the same and in yaml I thought I only would need to remove the esp8266 specific and change the pinnumbers to GPIO21.

however here it suggests that more changes are required
https://github.com/huizebruin/s0tool/issues/100#issuecomment-1385849075
https://github.com/huizebruin/s0tool/issues/125#issuecomment-1560091289
I already have to complete configuration running local without external components.

Am I missing something obvious here? or can I just go ahead and migrate to ESP32.
I think for wemos_d1 esp32 the v5 connection would be exactly the same as for esp8266

huizebruin, 2023-11-25:
The ESP32 is a microcontroller that operates with a supply voltage of 3.3V, and its GPIO (General Purpose Input/Output) pins are also designed to work with 3.3V logic levels. Applying a voltage higher than 3.3V to the GPIO pins can potentially damage the ESP32

bkbartk, 2023-11-26:
good to know,
but the board I'm talking about has some kind of power regulator
[image]
applying 5v to vcc should work, and also the locations of vcc, 3.3v and GND should be the same as for the other d1_mini

actually the pinout should be fully backward compatible with the d1_mini ESP8266, only the pinnumbers should sometimes be updated, and the is an additional row of pins which might prevent is from fitting in the case.

huizebruin, 2023-11-26:
The gpio pins cant handle 5v to there gpio , and the npn sensor sends 5v out when its getting a puls, and also needs 5v to work.

bkbartk, 2023-11-26:
Now I see, so if I would this, I need also to connect the npn to 3.3V,
Possibly also replacing the npn with one which support 3.3v

This will probably mean, I need to go full DIY and cannot use the one I ordered from the shop.

Interesting I thought the new board would be 100% backward compatible, but apparently not

I think for now I will live with the poor wifi reception.

If you ever decide to design an esp32 compatible equivalent I'm interested.

huizebruin, 2023-11-26:
Sadly there isn't a npn with one which support 3.3v.

u can think about mesh wifi so u have better reception.
like this.
<a href="https://partner.bol.com/click/click?p=2&t=url&s=1097464&f=TXL&url=https%3A%2F%2Fwww.bol.com%2Fnl%2Fnl%2Fp%2Ftp-link-deco-m4-mesh-wifi-3-pack-2018%2F9300000006179760%2F&name=TP-Link%20Deco%20M4%20-%20Mesh%20WiFi%20-%203-pack%20-%202018">TP-Link Deco M4 github</a> or this one  <a href="https://partner.bol.com/click/click?p=2&t=url&s=1097464&f=TXL&url=https%3A%2F%2Fwww.bol.com%2Fnl%2Fnl%2Fp%2Ftp-link-deco-x10-mesh-wifi-wifi-6-1500-mbps-3-pack%2F9300000162438858%2F&name=TP-Link%20Deco%20X10%20-%20Mesh%20WiFi%20-%20WiFi%206">TP-Link Deco X10 github bol.com</a>

bkbartk, 2023-11-26:
I have roaming wifi (with TP-Link Deco M4 v2 running openwrt), and the distance to the AP probably is not a problem,
however it can be improved, there are issues with small dropouts I only experience with esp82xx devices, and even not all of them.
it has something to to with openwrt ath10k and the esp82xx driver, and I tried all solutions I could find, nothing worked for me, so probably i need to replace my accesspoints, and I'm planning for that, but for now it's just much cheaper to replace a few esp controllers.

I will close this issue as my main question has been answered.

closed 2023-11-26 by bkbartk

bkbartk, 2023-11-26:
Maybe one question. 
would it be possible to add something like this
https://www.aliexpress.com/item/1005001621897460.html?src=google&src=google&albch=shopping&acnt=494-037-6276&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=19213834753&albag=&trgt=&crea=nl1005001621897460&netw=x&device=m&albpg=&albpd=nl1005001621897460&gad_source=1&gbraid=0AAAAADznYb9TdgIMCQBaC7qtG0P35yeGj&gclid=CjwKCAiA9ourBhAVEiwA3L5RFkug-5n8yEcDU6_jAP95rUL-TVqIMlEGOQ-rKARZjf3Z8WrekDkRBBoCv7UQAvD_BwE&gclsrc=aw.ds&aff_fcid=c512cd27948b41f49dbf6c95fa5b2c6b-1701003383177-05265-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=c512cd27948b41f49dbf6c95fa5b2c6b-1701003383177-05265-UneMJZVf&terminal_id=e4df6dff71fb4d079ab7308d56e48099&afSmartRedirect=y
to lower the voltage. 

Not planning this for now. But I’m just curious if it would be possible.

huizebruin, 2023-11-26:
No that won't work, because it is a pulse that u need to lower to the 3.3v

That pcb is only for power a tool on 3.3v.

Better to look at an opto in the tool but u must place more components on a Board.