huizebruin/s0tool #145: How to install gasenwater

PasekJ, 2023-12-09:
Hello, 
can you please provide information on how to set up gas and water measurement. 

Thank you very much for your support.

PasekJ, 2023-12-10:
Hello, 
please rename the file, the letter a is missing :-) Thank you

packages: [source /config/esphome/s0tool-standard.yaml:4]
  
   Failed to load packages. esphome/components/gaspuls.yml does not exist in repository.
   huizebruin.s0tool-standard: github://huizebruin/s0tool/esphome/gasenwater.yaml

huizebruin, 2023-12-10:
I will check it out, Thank you for your message.

huizebruin, 2023-12-10:
The gas sensor I can't test because we have a meter which is also a p1 and is sitting in the p1 electric meter.

It was a test to see if it would work also for other people.

If you're interested in testing it out for me. It would be nice.

PasekJ, 2023-12-10:
I don't have a gas meter P1 but s0, there is a tongue contact. I thought plynenwater was 2x s0 just with the right name.

 

S přátelským pozdravem

Jiří Pašek

mobil: +420 602 385 714
e-mail:  ***@***.***> ***@***.***
K Berounce 70, 267 18 Hlásná Třebaň, IČ: 19405031

commit a73f8233 (2023-12-21): Update gasenwater.yaml

Fix #145
closed 2023-12-21 by huizebruin