huizebruin/s0tool #150: ESPHome update issue - "Pin 4 is used in multiple places"

makkie2002, 2023-12-21:
After updating ESPHome to 2023.12.0 I get following compile error:

`INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/s0tool-watermeter-ad280f.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Brussels'
Failed config

sensor.pulse_meter: [source <unicode string>:66]
  
  Pin 4 is used in multiple places.
  platform: pulse_meter
  internal_filter_mode: PULSE
  internal_filter: 50ms
  pin: 
    number: 4
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  id: watermeter_flow
  name: watermeter flow
  state_class: measurement
  unit_of_measurement: l/min
  accuracy_decimals: 1
  icon: mdi:water-pump
  filters: 
    - lambda: !lambda |-
        return x * atof(id(Select_pulse_water).state.c_str()) * 1000;
  total: 
    id: sensor_pulse_meter_total
    name: watermeter Totaal
    state_class: total_increasing
    device_class: water
    icon: mdi:water
    unit_of_measurement: m³
    accuracy_decimals: 3
    filters: 
      - multiply: 0.001
    disabled_by_default: False
    force_update: False
  disabled_by_default: False
  force_update: False
  timeout: 5min
binary_sensor.gpio: [source <unicode string>:50]
  
  Pin 4 is used in multiple places.
  platform: gpio
  id: water_sensor_status
  name: Water sensor status
  internal: False
  pin: 
    number: 4
    inverted: True
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(total_water_pulses) += 1;
  disabled_by_default: False`

AppleFlapp, 2023-12-21:
The same results here. 

```INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/s0tool-standard.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Amsterdam'
Failed config

sensor.pulse_meter: [source <unicode string>:60]
  
  Pin 14 is used in multiple places.
  platform: pulse_meter
  pin: 
    number: 14
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  name: Actuele energie
  id: actual_power
  icon: mdi:flash
  accuracy_decimals: 0
  unit_of_measurement: W
  state_class: measurement
  device_class: power
  filters: 
    - lambda: !lambda |-
        return x * (60.0 / id(Select_pulse_rate).state)* 1000 ;
  total: 
    unit_of_measurement: kWh
    name: Totaal opgebracht
    id: W_opbrengst_totaal
    state_class: total_increasing
    device_class: energy
    icon: mdi:flash
    accuracy_decimals: 3
    filters: 
      - lambda: !lambda |-
          return id (id(total_kwh_pulses) * (1.0 / id(Select_pulse_rate).state) );
    disabled_by_default: False
    force_update: False
  disabled_by_default: False
  force_update: False
  internal_filter: 13us
  timeout: 5min
  internal_filter_mode: EDGE
sensor.pulse_meter: [source <unicode string>:66]
  
  Pin 4 is used in multiple places.
  platform: pulse_meter
  internal_filter_mode: PULSE
  internal_filter: 50ms
  pin: 
    number: 4
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  id: watermeter_flow
  name: watermeter flow
  state_class: measurement
  unit_of_measurement: l/min
  accuracy_decimals: 1
  icon: mdi:water-pump
  filters: 
    - lambda: !lambda |-
        return x * atof(id(Select_pulse_water).state.c_str()) * 1000;
  total: 
    id: sensor_pulse_meter_total
    name: watermeter Totaal
    state_class: total_increasing
    device_class: water
    icon: mdi:water
    unit_of_measurement: m³
    accuracy_decimals: 3
    filters: 
      - multiply: 0.001
    disabled_by_default: False
    force_update: False
  disabled_by_default: False
  force_update: False
  timeout: 5min
binary_sensor.gpio: [source <unicode string>:46]
  
  Pin 14 is used in multiple places.
  platform: gpio
  id: kwh_sensor_status
  name: kWh sensor status
  internal: False
  pin: 
    number: 14
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(total_kwh_pulses) += 1;
  disabled_by_default: False
binary_sensor.gpio: [source <unicode string>:50]
  
  Pin 4 is used in multiple places.
  platform: gpio
  id: water_sensor_status
  name: Water sensor status
  internal: False
  pin: 
    number: 4
    inverted: True
    mode: 
      input: True
      output: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(total_water_pulses) += 1;
  disabled_by_default: False

makkie2002, 2023-12-21:
There was some feedback from the ESPHome Github community: [https://github.com/esphome/issues/issues/5247](url)

[image]

huizebruin, 2023-12-21:
I will look at it tonight.

smarthomejunkie, 2023-12-21:
> There was some feedback from the ESPHome community: [https://github.com/esphome/issues/issues/5247](url).
> 
> ![image](https://private-user-images.githubusercontent.com/5372859/292220055-0a785cae-e7d6-43cb-840f-8aab0b217c3f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxNjY1NDAsIm5iZiI6MTcwMzE2NjI0MCwicGF0aCI6Ii81MzcyODU5LzI5MjIyMDA1NS0wYTc4NWNhZS1lN2Q2LTQzY2ItODQwZi04YWFiMGIyMTdjM2YucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjFUMTM0NDAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDNjZjcyN2E3N2VmOTE3NTc1NDViZjc3NWViMzNhNDUzNmUwY2UyMjExOWVlMjY1YTUxZTc1MGExZTA4MTcxMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.uYzkeWoN1Y7mJWSq5PzKuSFqg8nyfGUdlNG7N5tnu1w)

I've looked into the code, but I can't find any number: 4 in the code in the whole repository.

fmeus, 2023-12-21:
> [https://github.com/esphome/issues/issues/5247](url).

Search for D2, this is the pin constant

smarthomejunkie, 2023-12-21:
> > [https://github.com/esphome/issues/issues/5247](url).
> 
> Search for D2, this is the pin constant

Okay, created PR: https://github.com/huizebruin/s0tool/pull/151

AppleFlapp, 2023-12-21:
It's about pin 4 (D2) and pin 14 (D5).

huizebruin, 2023-12-21:
I'm fixing the other files to and also #152 thanks for the heads up and help @smarthomejunkie

JGeeforce, 2023-12-21:
Just wanted to report this too. Upgraded to 2023.12.1, also there an issue.

huizebruin, 2023-12-21:
https://github.com/huizebruin/s0tool/releases/tag/v23.12.34

should bee fixed by now.

closed 2023-12-21 by huizebruin