huizebruin/s0tool #156: add initial meter total in ESPhome

stingone, 2024-01-14:
Could you add the option to set the initial water meter value from the meter itself into your code?

i used the code below and it works fine.

`number:
  - name: Initiële meterstand m3
    platform: template
    id: initial_water_value
    device_class: water
    min_value: 0
    max_value: 100000
    step: 0.001
    entity_category: config
    icon: mdi:water
    unit_of_measurement: "m³"
    set_action:
      - lambda: id(total_water_pulses) = x * 1000;`

huizebruin, 2024-01-20:
Hello @stingone 

When i add this to my esphome config it stays on 0.001 and i used a lot of water today.

```yaml 
number:
  - name: Initiële meterstand m3
    platform: template
    id: initial_water_value
    device_class: water
    min_value: 0.001
    max_value: 100000
    step: 0.001
    entity_category: config
    icon: mdi:water
    unit_of_measurement: "m³"
    set_action:
      lambda: id(total_water_pulses) = x * 1000;  
      ```
      
 what did you else ?

stingone, 2024-01-20:
> Hello @stingone
>
> When i add this to my esphome config it stays on 0.001 and i used a lot of water today.
>
> ```yaml
> number:
>   - name: Initiële meterstand m3
>     platform: template
>     id: [...]

You should get an extra field called Initiële meterstand m3. You can than fill-in once the initiele meterstand and hit enter. It then updates the total total_water_pulses once to the entered value. 

After that the field returns to 0.

See my current code based on your code. 

`esphome:
  name: ${name}
  name_add_mac_suffix: false
  
substitutions:
  name: water-sensor
  device_description: ${name} gemaakt door huizebruin.nl versie ${vdate}.
  vdate: "v23.12.34" 
  date: "21-12-2023"

esp8266:
  board: d1_mini

preferences:
  flash_write_interval: 120min

improv_serial:

web_server:
  port: 80
  
# Enable logging
logger:
  level: INFO
  esp8266_store_log_strings_in_flash: false

ota:

time:
  - platform: homeassistant
    id: time_homeassistant
    on_time_sync:
      - component.update: S0tool_uptime_timestamp  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Water-Sensor Fallback Hotspot"
    password: "BRqxU0cJnwJg"

captive_portal:

# Delen en bewerken mag maar laat deze eerste regels staan.
# Bron : https://www.huizebruin.nl
# Bron File : https://github.com/huizebruin/s0tool/tree/main/esphome
# © Huizebruin.nl
#  # ⬇ Enable Home Assistant API ⬇ # 
api:
  services:
    - service: meterstand_water
      variables:
[... 155 lines cut ...]
    internal: False

#------------------------#  sensoren #------------------------#
  - platform: uptime
    id: S0tool_uptime

  - platform: template
    id: S0tool_uptime_timestamp
    name: "${name} online sinds"
    device_class: "timestamp"
    accuracy_decimals: 0
    update_interval: never
    icon: 'mdi:clock-start'
    entity_category: diagnostic
    lambda: |-
      static float timestamp = (
        id(time_homeassistant).utcnow().timestamp - id(S0tool_uptime).state
      );
      return timestamp;  

# ⬇ versie verbinding informatie sensoren voor in ha ⬇ #
  # WiFi signal
  - platform: wifi_signal
    name: "${name} - WiFi Signal"
    update_interval: 120s`

closed 2024-02-23 by huizebruin