huizebruin/s0tool #165: Does not work after rebooting

dychonn, 2024-03-09:
Hi, 

I just soldered an wemos d1 with an 5v npn sensor (from aliexpress) black wire to the d2 and blue to ground and brown to 5v. I have no resistor soldered between the brown and black wire at the moment. Because i read that it needs to work without. And it did a few times, and i tested before with the same problem after a few times rebooting. So the led is lighting up when i hold metal for the sensor. But when i look in my home assistant "water sensor status" stays off. That's why it is not registrering any water usage. The strange thing is it did work like this. My network is also saying that is online, but not giving any usage back in my home assistant.

dychonn, 2024-03-09:
Kan gesloten worden. De draad op d2 was blijkbaar niet goed gesoldeerd, waardoor deze zoals gedacht niks meet.

huizebruin, 2024-03-10:
Mooi dat je het gevonden hebt.

De weerstand is ervoor dat hij naar beneden getrokken wordt. En dus niet gaat zweven. En daardoor geen goede plus doorgeeft.

closed 2024-03-10 by huizebruin