huizebruin/s0tool #180: Problemen with update

huizebruin, 2024-06-26:
Problem with 2 times the platform ESPhome in the code.

I will try to fix it tonight.

[image]

@schmm  thanks for the testing.

Hurr1c4n3, 2024-06-26:
Also got a problem with the last update but it seems different:
[image]

```
INFO ESPHome 2024.6.3
INFO Reading configuration /config/esphome/s0tool-watermeter-c03608.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Amsterdam'
Failed config

update.http_request: [source /data/packages/b59da29a/esphome/components/s0tool-watermeter-update.yml:2]
  
  Component update.http_request requires 'platform: http_request' in component 'ota'.
  platform: http_request
  name: Firmware Update
  source: https://s0tool.nl/s0tool-watermeter-manifest.json
```

Hurr1c4n3, 2024-06-26:
After flashing the esp from the s0tool website and after that through the esphome on Home Assistant it works again.

huizebruin, 2024-06-27:
Should be fixed in https://github.com/esphome/issues/issues/5980 ESPHome version 2024.6.4 and above

huizebruin, 2024-06-27:
> After flashing the esp from the s0tool website and after that through the esphome on Home Assistant it works again.

Good to here it is working again.

commit 742fd9b3 (2024-07-08): Update http://s0tool.nl

* New site
* Cleanup html site
* Fixed : #180 thanks for your feedback and help  @schmm and @Hurr1c4n3
* Fixed : #203 thanks for your feedback and help @wes1993
closed 2024-07-08 by huizebruin