huizebruin/s0tool #203: Failed config during update

wes1993, 2024-07-06:
Update error, below the log:
```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/s0tool-standard-flux-5ca9c0.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Rome'
Failed config

button.template: [source /data/packages/b59da29a/esphome/components/basis.yaml:112]
  platform: template
  id: check_for_update_button
  name: Check for firmware update
  device_class: update
  on_press: 
    - then: 
        - component.update: 
            
            Couldn't find ID 'update_http_request'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "ota_http_request".
            id: update_http_request
  disabled_by_default: False
```

My config:
```
substitutions:
  name: s0tool-standard-flux-5ca9c0
packages:
  huizebruin.s0tool-standard-flux: github://huizebruin/s0tool/esphome/pulscounter_flux.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: Ij63ZgPKEUgNtLTvCsS2E2CvbWAvYYZxpRLs/YCEywI=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

```

Best Regards
Stefano

huizebruin, 2024-07-06:
Hello  Stefano,

Do you use the latest version of home assistant 2024.7.0?

wes1993, 2024-07-06:
Yes

wes1993, 2024-07-06:
[image]

wes1993, 2024-07-06:
Hello again @huizebruin,
Thanks for your support, from what I can see the problem seems related to this part of the code in basis.yaml:

```
  - platform: template
    id: check_for_update_button
    name: Check for firmware update
    device_class: update
    on_press:
      then:
        - component.update: update_http_request
```

Best Regards
Stefano

wes1993, 2024-07-06:
I think that the problem is:
Inside the basis.yaml file there is this part of the code:
```
  - platform: template
    id: check_for_update_button
    name: Check for firmware update
    device_class: update
    on_press:
      then:
        - component.update: update_http_request
```
But in the esphome/pulscounter_flux.yaml there isn't reference to esphome/components/pulscounter_flux-update.yml:
```
packages:
  remote_package:
    url: https://github.com/huizebruin/s0tool/
    ref: main
    files:
      - esphome/components/basis.yaml
      - esphome/components/kwhpuls.yml
      - esphome/components/watermeter_flux.yaml
```
So I think we should create the file esphome/components/pulscounter_flux-update.yml that access the Json of pulscounter_flux.yaml with updated version
For now I have fixed in this way but clearly the update part won't work because the website https://s0tool.nl/s0tool-standard-flux.json does not exist:

My updated ESPHome config:
```
substitutions:
  name: s0tool-standard-flux-5ca9c0
packages:
  huizebruin.s0tool-standard-flux: github://huizebruin/s0tool/esphome/pulscounter_flux.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: HiddenByMe

**update:
  - platform: http_request
    name: Firmware Update
    id: update_http_request
    source: https://s0tool.nl/s0tool-standard-flux.json**

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

```

huizebruin, 2024-07-07:
Hello Stefano,

I found the problem, i will try to fix this as soon as possible.

huizebruin, 2024-07-07:
Hello Stefano,

Can you tell me what for sensor you are using, i will put it also on the flashpage so everyone can use it from the flash page.

wes1993, 2024-07-07:
Hello @huizebruin,
Thanks for your help, sure I'm using this sensor:

Dekaim Flussostato Acqua, 1Pc... https://www.amazon.it/dp/B08KGG1H8Y?ref=ppx_pop_mob_ap_share

Best regards
Stefano

commit 742fd9b3 (2024-07-08): Update http://s0tool.nl

* New site
* Cleanup html site
* Fixed : #180 thanks for your feedback and help  @schmm and @Hurr1c4n3
* Fixed : #203 thanks for your feedback and help @wes1993
closed 2024-07-08 by huizebruin