huizebruin/s0tool #209: KWH meter stand niet op nul kunnen beginnen

BenLukken, 2024-07-11:
Beste Wobbe,

vraagje ik heb gisteren de S0Tool in HA gezet, nu heb ik een KWh meter die met ) kWh op de meter begint
echter in HA start hij met 8000 KWh, is dit ook ergens aan te passen ?
[image]

ik hoor het graag ( ps geen haast, de warmte pomp werkt nog niet )

Groetjes Ben

huizebruin, 2024-07-11:
Hallo Ben,

Mooi om te horen dat het gelukt is en dat hij werkt.

Hier heb ik beschreven hoe je de meterstanden kunt veranderen.

https://www.huizebruin.nl/wiki/s0tool-meterstand-invullen-in-home-assistant

En heb al meer gedeeld op die Wiki pagina.

En ook op mijn discord groep hebben we al een aantal leuke dingen gedeeld voor de S0tool en Home Assistant.

⁣Met vriendelijke groet,
Wobbe

https://www.huizebruin.nl
https://www.s0tool.nl

Wil je mee chatten met ruim 200 leden over de S0tool en 3d printen en elektronica en home assistant en meer. Kom dan gezellig langs op mijn discord groep. https://discord.gg/AskpfnY

Volg de site voor updates op social media.
Twitter https://twitter.com/huizebruin
Facebook https://www.facebook.com/huizebruin
Instagram https://www.instagram.com/huizebruin
Of schrijf je in voor de nieuwsbrief.
https://www.huizebruin.nl/nieuwsbrief
Rss feed http://www.huizebruin.nl/feed​

huizebruin, 2024-07-11:
Hallo Ben,

Hier heb ik dat beschreven in het Nederlands.
[https://www.huizebruin.nl/wiki/s0tool-meterstand-invullen-in-home-assistant/](https://www.huizebruin.nl/wiki/s0tool-meterstand-invullen-in-home-assistant/)
En hier heb ik nog wat meer geplaatst. [https://www.huizebruin.nl/wiki/s0tool/](https://www.huizebruin.nl/wiki/s0tool/)
and on [https://s0tool.nl](https://s0tool.nl) i worte it in English.

huizebruin, 2024-07-17:
@BenLukken  is het reeds gelukt ?

BenLukken, 2024-07-17:
Goedenavond Wobbe,
Jazeker het is gelukt met jouw jouw handleiding tip was het een eitje💪
Ik ben nog niet zo doorgewinterde programuur in home assistent langzaam
maar zeker kom ik er wel😂 nogal de toepassing voor twee extra
kilowattuurmeter meters.
Dus tzt nog 2 printje nodig, of is het mogelijk om meer meters op 1 printje
aan te sluiten?

Groetjes Ben

closed 2024-08-25 by github-actions[bot]

RonaldRegister, 2025-01-07:
same issue as #236  ?