huizebruin/s0tool #210: High spikes in solar kWh meter measurement

RobPijpers, 2024-07-23:
My brand new s0tool periodically shows very high spike spikes in the measurement graph in Home Assistant, see screenshot below.
I already replaced the power supply and the s0tool is running the most recent firmware.
What could be the cause of this?

[image]

RobPijpers, 2024-07-23:
It looks like the spikes are increasing today
[image]

huizebruin, 2024-07-23:
Can you give me a little more information. 

Did u build it by your own? 
Or did you buy the s0tool?

And which software you are using,  and can you tell me witch software u are using. 
Standard or kwh or special?

RobPijpers, 2024-07-23:
I bought the complete set with kWh meter software installed version 3.6.1

RobPijpers, 2025-08-17:
@huizebruin, I was wondering why this issue is closed as there is no solution if I'm not mistaken?
Anyhow, I managed to filter the spikes previously and currently moved to a more stable solution.

huizebruin, 2025-08-17:
I accidentally closed this, and I haven't been able to fix it yet