huizebruin/s0tool #214: issue with your new update button of 240703?

RealDoublejan, 2024-08-03:
[image]

i can't update the watermeter, it seems your new update button throws an error. 
ESPHome is 2024.07.03 on my side.

huizebruin, 2024-08-05:
Can u connect the (s0tool) wemos to your pc or laptop  and go to [https://s0tool.nl/Installation.html](https://s0tool.nl/Installation.html)
and flash it with the s0tool watermeter ?

To test if this is the sollution.
you dont have te delete the Home assistant and esphome devices for the s0tool.

RealDoublejan, 2024-08-05:
Hey,
I went with a laptop to my cellar to do a download as you suggested, which failed bin error 404. So then I just tried to validate it again in earphone and it went through. Also my kWh meter was able to be upgrade…. I felt really stupid for a few minutes until I saw you did an update to your basis.yaml file… 
Thanks a lot. This issue can be closed

closed 2024-08-05 by RealDoublejan