huizebruin/s0tool #234: stream pointer vanished

bert1111, 2024-09-28:
**Describe the bug*
I had some wifi issues (broken antenna) which were resolved but suddenly the s0 meter vanished from my network.
I installed the latest update through the website with a usb cable.
I can set the network and then after a few seconds I see stream pointer vanished. And device connection is lost again.
I did find that there is a bug somewhere with esp home but not yet solved. Is there a workaround? Strange thing is that it has worked for two months before the wifi problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

huizebruin, 2024-09-28:
Do you see a ssid with the name ```S0tool***``` with your mobile?
Than its not connected to your wifi network.

And than you must connect it again to your network.
you can connect to the s0tool ssid.
and go to ```192.168.4.1``` and there you can connect to your wifi network again.

bert1111, 2024-09-28:
no that ssid doesn't appear anymore. it has something to do with this issue. https://github.com/esphome/issues/issues/5949. I don't know if you're aware of this.

huizebruin, 2024-09-28:
What i heart it was fixed an little time ago. But it doesn't when i see this.
i will contact some of my other Dutch users on my discord if they also have the same error after the update or flashing the s0tool.

What version of Home assistant and ESPHome are you using?

bert1111, 2024-09-28:
Firmware: 3.6.2 (ESPHome 2024.9.1)
Versie | core-2024.9.3
Installation type | Home Assistant OS
What does appear to work is not finish the flashing by skipping the search for wifi networks.
Than the s0tool network appears. Go to the url you mentioned above.
Set the wifi there. The watermeter is now connected for a few minutes were it previously disconnected after a few seconds with mentioned error.

bert1111, 2024-09-28:
Nope. It's gone again. Same error [http_request.arduino:137]
Stream pointer vanished!

huizebruin, 2024-09-28:
When u use the same esp it stores the ssid and password in his esp.
That's why you don't see the selecting ssid (wifi) list.

bert1111, 2024-09-28:
I think I'm not explaining it good enough. Or I don't understand it enough.
I'll try again but be clearer of the steps I'm taking.

So the watersensor was working fine up until last night with version 3.6.1.
Today I saw in HA that all watermeter sensors were not available.
I reinstalled via the webpage to version 3.6.2.
And from the moment I select my wifi network I lose connection with error http_request.arduino:137]
Stream pointer vanished!

The last thing I tried was reinstalling via the website and skipping the set wifi part.
Then the s0*** network appears. 
I chose that network and set the wifi via 192.168.4.1 like you said.
Then the s0 network disappears because it is connected to my home wifi. So I do see my ssid to connect with in the list.

After that, it is connected and is visible again on my network until it starts giving the http_request.arduino:137]
Stream pointer vanished! error. 
This was the case first after a few seconds with the first way. 
And with the skipping the set wifi part, but setting through the 192.168.4.1 after a few minutes.

bert1111, 2024-09-28:
This is the log after succesfully connecting to my wifi.
And after this it even appears the device crashes after waiting a while. Page does not respond 

`[I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[I][captive_portal:032]: Captive Portal Requested WiFi Settings Change:
[I][captive_portal:033]:   SSID='xxxxxx'
[I][captive_portal:034]:   Password='xxxxxxxx'[redacted]
[W][component:157]: Component wifi set Warning flag: scanning for networks
[I][wifi:540]: - 'xxxxxxx
[I][wifi:313]: WiFi Connecting to 'xxxxx'...
[I][wifi:617]: WiFi Connected!
[W][component:170]: Component wifi cleared Warning flag
[W][component:170]: Component api cleared Warning flag
`

huizebruin, 2024-09-28:
I will look if i can remove the http sensor for now

scns, 2024-09-28:
I hope that Will fix the issue.

bjornpri, 2024-09-28:
Identical problem here on a Wemos D1 Mini V4.0.0 Type-C USB. Has been working fine for a long time. Until 3.6.2: Works after install, stops working with same issue explained in this thred after power recycle.

huizebruin, 2024-09-28:
i'm pushing now a version online without the ota update and http update .
https://github.com/huizebruin/s0tool/actions/runs/11085402377

huizebruin, 2024-09-28:
https://github.com/huizebruin/s0tool/actions/runs/11085452651
this looking better

huizebruin, 2024-09-28:
@bjornpri  and @bert1111 
Would you try it out to flash it from https://s0tool.nl

I uploaded the changes. 

https://github.com/huizebruin/s0tool/actions/runs/11085452651/job/30801987513

bjornpri, 2024-09-28:
I tried twice. No change. Is there a visual change I can look for so we know I have the new version?

huizebruin, 2024-09-28:
I named the version inside of the basis file 3.6.3

bjornpri, 2024-09-28:
Is still says 3.6.2 every where. I tried a third time. No change. After install i visit the device and it loks good. Then without removing power i press the restart link in the web UI and it comes up with the sam issue.

scns, 2024-09-28:
The release build is not yet started. @huizebruin, van you publish 3.6.3?

scns, 2024-09-28:
> I named the version inside of the basis file 3.6.3

I think that is not working because The firmware on the website is not updated on The site.

huizebruin, 2024-09-28:
Its building right now

scns, 2024-09-28:
Build is done!

scns, 2024-09-28:
@bjornpri @bert1111 can you try again?

huizebruin, 2024-09-28:
It should now be online,  i flasht a spare part one now from the website to 3.6.3.

bjornpri, 2024-09-28:
Almost. It survives all kinds of resets. but... I can not add it to ESP Home:

INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/s0tool-watermeter-202f4c.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Oslo'
Failed config

update.http_request: [source /data/packages/b59da29a/esphome/components/s0tool-watermeter-update.yml:2]
  
  Component update.http_request requires 'platform: http_request' in component 'ota'.
  platform: http_request
  name: Firmware Update
  id: update_http_request
  source: https://s0tool.nl/s0tool-watermeter/manifest.json

scns, 2024-09-28:
> Almost. It survives all kinds of resets. but... I can not add it to ESP Home:
>
>
>
> INFO ESPHome 2024.9.1
>
> INFO Reading configuration /config/esphome/s0tool-watermeter-202f4c.yaml...
>
> INFO Updating [...]

Dus you remove your old entry in esphome?

huizebruin, 2024-09-28:
im adding it to my beta machine.
[image]

bjornpri, 2024-09-28:
It failed while verifying the yml file with the error:
Component update.http_request requires 'platform: http_request' in component 'ota'.

I added it in ESPHome and I broke it again. 
I'll try once more deleting old stuff and restarting HA.....

It autodiscovers and I can 'adopt' it... I adopt it and I get:

[image]

huizebruin, 2024-09-28:
I've been testing it now a couple of times.  But it's working here now good and no losses that the device is going offline.

huizebruin, 2024-09-28:
Adopt first to ESPHome and after that to Home assistant. 

But did you delete it from your esphome dashboard also?

scns, 2024-09-28:
I think The repo needs to be refreshed on your'e instance. 

In the worst case you have to wait till tomorrow. (24 hrs)

I had with my own projects The same. 

Sounds weird but sometimes you have to wait. The device in the integration is still working if it has the same name

bjornpri, 2024-09-28:
> Adopt first to ESPHome and after that to Home assistant.
> 
> But did you delete it from your esphome dashboard also?

That is what I did. Deleted every thing i HA and ESPHome. It fails validating the yaml while adopting in ESPHome. I states it is missing OTA.

bjornpri, 2024-09-28:
> im adding it to my beta machine. [...]

ESPHome 2024.8.1? Im using 2024.9.1.

huizebruin, 2024-09-28:
[image]
or try clean build files , then validate , then install

bjornpri, 2024-09-28:
While selecting "Clean Build Files":

[image]

scns, 2024-09-28:
Just wait at least 24 Hours 😅

bjornpri, 2024-09-28:
It seems that ESPHome 2024.9.1 requires the platform: http_request to be able to properly adopt the device. I can controll it by setting new values and restart it from HA but the enteties in HA will not update. The API wil probably not work without he API encryption key...? Yes?

scns, 2024-09-28:
I Will test tomorrow.

bjornpri, 2024-09-28:
> I Will test tomorrow.

Thanks

huizebruin, 2024-09-28:
Me to and hopefully we find a solution for the problem.

scns, 2024-09-29:
[image]
[image]
[image]
[image]

I flashed this morning my production S0 tool. And everyrhing worked fine. 

Import in ESPhome, no problem. And the device came online in HA. I Will monitor of the device stay's online.

bjornpri, 2024-09-29:
Im testing this morning. Same issue.

[image]

My tool's yaml file seems to be missing platform: http_request' in component 'ota'.
@scns does your tools yaml file contain the ota component?

[image]

huizebruin, 2024-09-29:
@bjornpri  at at line 4 after yaml@main 
maybe is that it.

bjornpri, 2024-09-29:
@huizebruin What at line 4? Where?
Line 4 in the yaml file there is no ..@main
In the output log there is the timezone on both @scns and mine after ...@main.

huizebruin, 2024-09-29:
Do you flash it by this link https://s0tool.nl/Installation.html ?
or do you hit the install button inside the esphome dashboard ?

JGeeforce, 2024-09-29:
Yesterday at 23:23 S0tool was suddenly available. Long story short, this morning I clean build files, deleted the device in ESPhome and did a new install by connecting it to the USB port. It's back again.

bjornpri, 2024-09-29:
Yes. I use this: 
[image]

bjornpri, 2024-09-29:
> Do you flash it by this link https://s0tool.nl/Installation.html ? or do you hit the install button inside the esphome dashboard ?
> 
> remove the s0tool out of your esphome dashboard and than flash it .

Done this 3 times now. No sucsess

huizebruin, 2024-09-29:
[image]
im testing that one at his moment too

bjornpri, 2024-09-29:
The flash is good. Works fine. Does not breake the device. Stands reboots. All fine so far.

bjornpri, 2024-09-29:
@huizebruin 
Does your ESPHome yaml file for your device include:

ota:
  - platform: esphome
  - platform: http_request

huizebruin, 2024-09-29:
[image]
its adopted here
And still online for 15 min
[image]
And now a hard reboot of the device and its come normal online .
[image]

bjornpri, 2024-09-29:
> @huizebruin Does your ESPHome yaml file for your device include:
> 
> ota:
> 
> - platform: esphome
> - platform: http_request

@huizebruin Can you please confirm this?

bjornpri, 2024-09-29:
My device D1 mini v4 usbc brakes back to the original problem:
[image]

huizebruin, 2024-09-29:
no i did delete that for now @bjornpri 
its not necessary for esphome to need that.
and there are a lot more people that are having trouble with that at this moment for also other tools.

that's why i push version 3.6.3 online

huizebruin, 2024-09-29:
https://discord.com/channels/429907082951524364/1283861358483603456

we had the same problem before.
it looks that the package updates once every 24H.
i am looking if i can change that in my esphome config for the s0tool.
can you try it again later today ?

bjornpri, 2024-09-29:
It sems to work now. But I can not adopt it. If I do it fails as decribed above.
[image]

scns, 2024-09-29:
Try the adoption this evening or tomorrow. I think it Will work after Some time.

bjornpri, 2024-09-29:
> https://discord.com/channels/429907082951524364/1283861358483603456
> 
> we had the same problem before. it looks that the package updates once every 24H. i am looking if i can change that in my esphome config for the s0tool. can you try it again later today ?

I will try in a day or two remotely. My install is at a remote location and im leaving here and will not be back for a couple of weeks, But I will check back.

bert1111, 2024-09-30:
Hi, I just installed version 3.6.3. No error so far. I noticed that after I set the value of my watermeter again and then unplugged the usb and switched to a usb adapter power source, the value I just set was changed to the initial value of 8m³ again. I set it again and check later if it stays set correctly.

JGeeforce, 2024-09-30:
> Hi, I just installed version 3.6.3. No error so far. I noticed that after I set the value of my watermeter again and then unplugged the usb and switched to a usb adapter power source, the value I [...]

Noticed the same behavior here.....

[3 more replies]

closed 2024-11-17 by github-actions[bot]