huizebruin/s0tool #237: Failed to build after ESPHome update

paganl, 2025-01-17:
### Please verify that this bug has NOT been raised before.

- [x] I checked and didn't find a similar issue

### Describe the bug*

After the latest update to ESPHome (HA), it no longer complies. 

INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/s0tool-standard.yaml...
INFO Updating https://github.com/huizebruin/s0tool/@main
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing s0tool-standard (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
|-- ESP8266HTTPClient @ 1.2
|-- noise-c @ 0.1.6
Compiling .pioenvs/s0tool-standard/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/dashboard_import/dashboard_import.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/factory_reset/button/factory_reset_button.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/http_request/http_request.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/http_request/http_request_arduino.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/http_request/http_request_idf.cpp.o
Compiling .pioenvs/s0tool-standard/src/esphome/components/improv_base/improv_base.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/s0tool-standard/src/esphome/components/http_request/http_request_arduino.cpp.o] Error 1
========================= [FAILED] Took 23.13 seconds =========================

Any help would be greatly appreciated.

Many Thanks

### Expected behaviour

To compile 

### Version Information

current, it should be pulling it from the repo; I've tried a clean build. 

### ESPHome version Information

INFO ESPHome 2024.12.4

### Home Assistant version Information

Core: 2025.1.2
Supervisor: 2024.12.3
Operating System: 14.1
Frontend: 20250109.0

### Additional context

None

scns, 2025-01-17:
My Build went successful: 

[image]

paganl, 2025-01-17:
Maarten, thanks! You encouraged me to look further. I reinstalled esphome, rebooted, and cleaned the build. After that, the build succeeded, so it seems there was a mismatch somewhere.

Many thanks

Paul

closed 2025-01-17 by paganl