huizebruin/s0tool #252: timeout: 15s is too small to detect leaks

samimasad, 2025-05-27:
### Please verify that this bug has NOT been raised before.

- [x] I checked and didn't find a similar issue

### Describe the bug*

consider to make this value much bigger, I was using this to detect water leak with small value over very lengthy time but now it doesnt work , at min it can detect 1 L every 15 seconds , so minimum of 4L/m , i would think you got to make it 300 seconds

### Expected behaviour

you could at least try to make it variable if someome want to deviate from your default

### Version Information

s0tool 3.6.4 , HA 2025.05.3 , esphome [2025.4.2]

### ESPHome version Information

2025.4.2

### Home Assistant version Information

2025.5.3

### Additional context

2025.05.3

huizebruin, 2025-05-27:
I will test it out here,
to look if this is working as needed.

samimasad, 2025-05-28:
tested it with value of  timeout: 700s , that us theory should allow for flow as little as 60/700 = around 0.1 L/min , that works fine for me , if this is ok i can push this small change

huizebruin, 2025-05-29:
here the same outcome, and it works better.

samimasad, 2025-05-29:
I will push the change