huizebruin/s0tool #253: Wrong watermeter stand after update.

Benvdaa, 2025-05-28:
### Please verify that this bug has NOT been raised before.

- [x] I checked and didn't find a similar issue

### Describe the bug*

When using service to update the correct watermeter stand 363,851 m3 water will be 0,364 m3 water.
Why using the pulserate  (0.001) on the watermeter stand update service ? 

### Expected behaviour

  services:
    - service: meterstand_water
      variables:
        meter_value: float
      then:
        - globals.set:
            id: initial_water_usage
            value: !lambda "return ( meter_value * atof(id(Select_pulse_water).state.c_str()) *1000 ) ;"

### Version Information

Firmware: 3.6.4 (ESPHome 2025.5.1)
Hardware: 3.6.1

### ESPHome version Information

ESPHome 2025.5.1

### Home Assistant version Information

Core            2025.5.3
Supervisor   2025.05.3
Operating System  15.2
Frontend     20250516.0

### Additional context

Watermeter stand update service showing m3 as L valvue

huizebruin, 2025-05-28:
I die this because you can have different pulses. 

But it's not working very well. 
And i don't can't find it out how to solve it.  At this point. 

If you know something to repair it. 
Please help me with it.

Benvdaa, 2025-05-30:
I did a quick solve bij putting a multiply 1000 in the formula
value: !lambda "return ( meter_value * atof(id(Select_pulse_water).state.c_str()) *1000 ) ;"

 363.851  * 0.001  = 0.364 m3

363.851 * 0.001 * 1000 = 363.851 m3

Benvdaa, 2025-08-17:
Sorry Wobbe,
Pulsrate water per m3 0.001
Watermeter stand before update 368,830 m3 
Watermeter stand after update 368,826 m3.
Correction with Calls the service meterstand_water of the node s0tool-standard  368,831 m3
Result: Watermeter stand 0,369 m3

With this correction in the api service it is working correctly for me.

[s0tool-standard.yaml](https://github.com/user-attachments/files/21822674/s0tool-standard.yaml)

huizebruin, 2025-08-17:
@Benvdaa thanks for sharing.
i will look at it and test it here , if its working good i will add it to the s0tool code.

Benvdaa, 2025-08-17:
I believe that the issue is related with the calculation with the pulserate in the service call.
The value  368,831 m3 * 0,001 = 0,368831  rounded 0,389 m3.
To elliminate the pulserate, I multiply it with 1000.