huizebruin/s0tool #264: Updated from 3.6.3 to 3.6.6: m3 instead of liter for watermeter totaal

JGeeforce, 2025-08-10:
### Please verify that this bug has NOT been raised before.

- [x] I checked and didn't find a similar issue

### Describe the bug*

Yesterday I updated from 3.6.3 to 3.6.6 and now I see in that the watermeter totaal shows m3 instead of liters like before. Was that a change? How can I change it back to liters? 

This is in the energy dashboard!

[image]

[image]

### Expected behaviour

I would  like to see liters again.

### Version Information

3.6.6

### ESPHome version Information

2025.8.0

### Home Assistant version Information

2025.8.0

### Additional context

N/A

JGeeforce, 2025-08-10:
Besides that, I see very strange things in the graphs when I go beck in time on the energy dashboard:

[image]

huizebruin, 2025-08-10:
## Not a S0tool bug — change in Home Assistant 2025.8.x

This is caused by a change in the **Home Assistant Energy Dashboard** in version `2025.8.x`, not by S0tool `3.6.6`.  
Starting in 2025.8.x, the Energy Dashboard **enforces `m³`** as the only allowed unit for water totals  
(`device_class: water` + `state_class: total_increasing`).

Previously, you could provide liters (`L`) and it would still work in the Energy Dashboard.  
Now, HA automatically expects and displays `m³`.

---

### S0tool default config
This is the **standard configuration** the S0tool has always used:

```yaml
unit_of_measurement: "m³"
filters:
  - multiply: 0.001
```
This is correct for the Energy Dashboard, which now requires cubic meters.

Want liters in your own dashboards?
If you also want to show liters for other views in Lovelace,
you can add a template sensor:

```yaml
template:
  - sensor:
      - name: "Watermeter Totaal (L)"
        unit_of_measurement: "L"
        state: "{{ (states('sensor.watermeter_totaal') | float * 1000) | round(0) }}"
        state_class: total_increasing
        device_class: water
```
Use the m³ sensor in the Energy Dashboard.

Use the liters sensor in your own dashboards or cards.

Conclusion: This is an HA core change in 2025.8.x to standardize units,
not a regression in S0tool.

JGeeforce, 2025-08-10:
Thanks. Missed that one in the update for HA.

huizebruin, 2025-08-15:
**Update**:
The S0tool configuration now also includes the total water meter in liters alongside the version in m³ by default.
Both values will appear automatically in Home Assistant without any additional setup.

https://github.com/huizebruin/s0tool/actions/runs/16987207329
See also: https://github.com/huizebruin/s0tool/issues/264